kwindrem/RpiDisplaySetup #1: Display timeout not working, dimmer not appearing

CzarofAK, 2021-05-07:
I encountered the following issue:
- The display time does not work, stays always on
- does not show the dimming setting
Firmware installed: v2.70 5 large 18

    SignalK = on
    Node-Red = off

kwindrem, 2021-05-07:
There is an option in the setup script to enable the dimming or not. This only works for the Raspberry PI 7" touchscreen as there is no mechanism to dim via HDMI.

The time should work, but the display needs to be connected to the first HDMI display. I probably need to add a setup option to set which display if people are connecting to the second HDMI display port on the PI. Let me know.

CzarofAK, 2021-05-07:
I use the standard raspberry pi 7" touchscreen.
during the installation, i have selected the dimmer option.

kwindrem, 2021-05-07:
Sounds like it should work. It's working for me with that display but I haven't checked that specific Venus version. I'll do that tomorrow.

You do need to reboot after installing.

CzarofAK, 2021-05-07:
i deinstalled now your "RpiDisplaySetup" and tried it with the manual of him:
https://obenschlaefer.com/low-budget-energiemonitor-victron-venus-os-mit-raspberry-pi-obenschlaefer/#Backlight_Overlay_installieren
there the display timeout works, but the dimmer bar is also missing.
check "Achtung – Kleines Update vom 17.04.2021:" it seems it has to do something with the Venus after v2.60

waiting for your response tomorrow.
thanks a lot for your commitment!

kwindrem, 2021-05-07:
Checked large-18 on my RPI 4 and both display dimming and blanking work. So now we need to figure out why it is not working for you.

Run my setup script and choose uninstall. Then run it again and choose install. This will insure all the mods are in place.
Watch the console while running the script and note any errors.

If you still have problems, check the following:

cat /etc/venus/backlight_device
This file should contain: /sys/class/backlight/rpi_backlight
The presence of this file will control whether the dimming function appears on the Display setup page
It's contents directs the Venus GUI to the device it needs to control to set the dimming level

cat /etc/venus/blank_display_device
This file should contain: /sys/class/graphics/fb0/blank
The presence of this file will control whether the display timeout function appears on the Display setup page
It's contend directs the Venus GUI to the device it needs to control to blank the screen

Other instructions will probably say the file should contain /sys/class/backlight/rpi_backlight/bl_power
I found that by changing it to fb0 blanking will also work on HDMI displays. However, there are two HDMI ports and two frame buffers, so changing this to fb1 would blank the second HDMI port.

Check to see if the following files exist:
  /u-boot/overlays/rpi-backlight-overlay.dtb
  /u-boot/overlays/rpi-display-overlay.dtb
My script installs the first one but the Venus images include the second one

Verify /u-boot/config.txt should contains hdmi_blanking=1:

grep hdmi_blanking  /u-boot/config.txt

My script edits config.txt to add this so it's a check to verify the file was edited properly.

I have checked files in the boot partition and they all match the ones on my system except for config.txt and cmdline.txt.
I modified cmdline.txt for other purposes that would not affect the display
I hand modified config.txt for my system and my setup script also modifies it so those are expected.

kwindrem, 2021-05-07:
I have modified the setup script to ask for a display type and port: RPI (DSI) display, HDMI port 0 and HDMI port 1. This may be helpful in debugging installation issues.

The RPI selection enables display dimming. An HDMI selection disables it since the HDMI protocol does not support dimming.

I also switched the blanking to use the RPI device instead of fb0 in case that's an issue on your system.

Let me know which PI you are using. All my testing has been on the RPI 4 v1.1 and v1.2 boards. I do know that RPI 4 v1.4 boards don't boot Venus OS (yet).

Let me know if you are still having issues or if you get it working. Especially let me know if you get it working and need to modify something the script does so I can accommodate it.

CzarofAK, 2021-05-08:
I use a RPI 3B going to change it to a 3B+ next week, as soon as it arrives.
- The time out works now.
- The dimmer bar is still not visible:

- Did not show any errors while carrying out the installation
- cat /etc/venus/backlight_device = is there
- cat /etc/venus/blank_display_device = "bl_power" i guess thats to you new changes
- /u-boot/overlays/rpi-backlight-overlay.dtb = is there
- /u-boot/overlays/rpi-display-overlay.dtb = is there
- /u-boot/config.txt should contains hdmi_blanking=1: = yes it does

kwindrem, 2021-05-08:
I'm glad the timeout is working.

Yes, the change in blank_display_device content was a change I just made. I did that because I thought the use of fb0 might be different on RPIs other than the 4.

Not sure why dimming isn't appearing. 

/etc/venus/backlight_device file must contain /sys/class/backlight/rpi_backlight. If this file does not exist or is empty, this could explain why the dimmer doesn't appear in the GUI, but my script should make all the necessary modifications.

I have a PI that uses an external HDMI display, not the "Official Raspberry Pi 7" Touchscreen". On that system, the dimming bar never appears in the Display menu. So the logic Venus uses doesn't just look for the presence of /etc/venus/backlight_device.

CzarofAK, 2021-05-09:
/etc/venus/backlight_device file contains /sys/class/backlight/rpi_backlight
i have deleted the folder /data/RpiDisplaySetup-current and /data/RpiDisplaySetup
then i have carried out your installation procedure again
"wget -qO - https://github.com/kwindrem/RpiDisplaySetup/archive/current.tar.gz | tar -xzf - -C /data
mv /data/RpiDisplaySetup-current /data/RpiDisplaySetup"

may do i have a issue with the rpi backlight installation? like this guy here?
https://community.victronenergy.com/questions/56217/venus-os-v26034-available-for-testing.html?childToView=56411#comment-56411

going to fetch a new Venus OS image, and start from scratch. I guess i messed someting up myself.
Let you know

5teveO, 2021-05-13:
Just to add to this.. I'm using a 5" DSI screen that works great, screen blank is there.. but no dimming bar.. 

All checks above are coming in exactly as CzarofAK findings.. i followed some more instructions elsewhere but they didnt help either.. 

Raspberry PI 3b+ and latest 'normal' version 2.70~11

Thanks

Steve O

CzarofAK, 2021-05-13:
Today i have flashed a new SD card with the "venus-image-large-raspberrypi2-20210416104645-v2.70~5-large-18" image.
Installed the SD card in my RaspberryPi3B+ with attached 7" Raspberry Touchscreen.
Did no additional changes than setting Wifi up, language, SSH and timezone.

running your scripts:
"wget -qO - https://github.com/kwindrem/RpiDisplaySetup/archive/current.tar.gz | tar -xzf - -C /data
mv /data/RpiDisplaySetup-current /data/RpiDisplaySetup"
and
"wget -qO - https://github.com/kwindrem/SetupHelper/archive/current.tar.gz | tar -xzf - -C /data
mv /data/SetupHelper-current /data/SetupHelper"

then i tried "/data/RpiDisplaySetup/setup":
root@raspberrypi2:~# /data/RpiDisplaySetup/setup
wc: /var/log/SetupHelper: No such file or directory
/data/SetupHelper/LogHandler: line 68: [: -gt: unary operator expected

seems like there is a file missing.

CzarofAK, 2021-05-13:
then i have installed it with "i" and "R" for the Raspberry Display.
After the restart, the display timeout appeared and is working. But the dimmer is not even shown in the menu.

5teveO, 2021-05-13:
I updated to the latest normal release from the large version to see if that affected anything.. it didnt.. 

Steve

kwindrem, 2021-05-13:
I will look into the log file error. I'm pretty sure it happens when the log file has not been created yet.

I was looking at the code and remembered someone else had an issue with dimming not appearing. In his case, AutoBrightness was set.

run dbus-spy and go into com.victronenergy.settings

scroll down to find Settings/Gui/AutoBrightness

If the value is 1, select this line, hit enter, 0, enter

I don't know why AutoBrightness is set. It should only be set for the Cerbo/Touch 50, but maybe the later versions of Venus OS have initialized this to true.

I will add code in setup to make sure it is set to false.

kwindrem, 2021-05-13:
I have confirmed AutoBrightness is being set when default settings are created!!!!!

I will patch my code, but in the meantime enter the following command:

dbus -y com.victronenergy.settings /Settings/Gui/AutoBrightness SetValue 0

CzarofAK, 2021-05-13:
Got it working! Greate Job!

closed 2021-05-13 by CzarofAK

5teveO, 2021-05-13:
Can confirm that sorts the dimming out here too.. is this meant to dim after a time also or is this just literally a set brightness slider?

kwindrem, 2021-05-13:
Glad things are working for you both. Sorry for the delay in locating the cause.

The dimming slider sets a fixed display brightness. It does not dim over time. 

The screen saver blanks the display abruptly at the given time of inactivity.

It would be nice if the two interacted and the screen dimmed gradually. Unfortunately, that would require modifying code I don't think we have access to (no source, and/or buried in the kernel).

5teveO, 2021-05-13:
Hi 

No apologies needed at all.. I appreciate you making this so painless! to confirm this DSI screen works perfectly.. https://www.aliexpress.com/item/33024647905.html?spm=a2g0s.12269583.0.0.5c4379dfnwlYdz 

Yes it would be nice if there was a dimming period.. say after 10 minutes dim to 25% and then after 30 minutes blank.. but hey.. i'm happy i've go this far :)

thanks again for your work..

Steve