kwindrem/RpiDisplaySetup #2: Default Brightness at power on

pete10004x4, 2021-07-28:
Rpi 3B+ running 2.72
My RP 7" display dims to 0 just below 50 and at reboot the display is too dim to see. Running remote console the Brightness bar is around 30% and I can move it up to 50% and the displays comes on normal brightness.
It remains at that brightness until the next power cycle. 
Should the Brightness be permanent, if not where does the brightness value get stored? I need to increase it to 50%

kwindrem, 2021-07-28:
Yes, brightness is persistent through reboots. It's stored with other persistent settings in /data/conf/settings.xml. 

I am running a RP 7" display on a RPI 4 on v2.72 without issues. My display dims smoothly from 100% all the way to about 10% (too dim to really see) then the display blanks.

I have not tested this on an PRI 3B+ but I don't see any reason it would not work.

RpiDisplaySetup installs the display dimming overlay and sets up a couple of pointers in /etc/venus then steps out of the way. Everything else is handled by the OS itself. There are no mods to the GUI files with this package. 

You can inspect and change the actual value with the  command line tool dbus-spy.
Brightness is com.victronenergy.settings /Settings/Gui/Brightness.

It sounds like all the software bits and pieces are installed properly or you wouldn't get any dimming.

Is it possible you have a defective display? Or a cable isn't seated completely? You might try installing Raspberry PI OS and see if things work there.

pete10004x4, 2021-07-28:
[image]
This is the screenshot of the display brightness whenever I reboot, looking at Settings the brightness is set to 65 however the screen does not reflect that.
Thanks for your comment, I think I will reload 2.72 and start with a clean install 
It works with a Pi OS

kwindrem, 2021-07-28:
I see the issue. The Settings parameter is limited to 100 while while the display's max brightness is actually 255.

The slider updates the display to it's value based on it's max value (slider at full right is 255) and that value lingers until reboot. But the max value stored in the Settings parameter is 100, so when the system reboots, it sends that value to the display That's about 40% max brightness.

I fixed this by resetting the dbus settings parameter to accommodate a range of 0 - 255.

pete10004x4, 2021-07-28:
Thank you that sorted it.

pete10004x4, 2021-07-28:
Still doesnt dim though, just reboots to a brightness level that turns the screen on. I am a bit perplexed as it did dim when I was using it originally on vers 2.58 I think but I went for a headless solution then

kwindrem, 2021-07-28:
I discovered a couple of bugs in my code that modifies the setting limits. So you probably didn't see a change. Try another update from GitHub. Sorry. I rushed out the change.

pete10004x4, 2021-07-29:
I saw that Brightness max had been set to 255 on update however on checking settings.xml after reboot I get this:
 <Brightness type="i" min="0" max="100" default="100" silent="False">150</Brightness>
the 150 is what I have to put in to get my display to switch on after reboot, on first reboot after update this was set to 79. I have tried to manually change the max and the default values but after reboot they always come back to 100. The only thing that doesnt change is the 150.

I think there is a bug in 2.72 preventing display dimming on a 3B+ as I'm not the only one who has this issue.

kwindrem, 2021-07-29:
I think it's fixed now. I discovered that the gui code resets the max value to 100 when it starts up. So I've modified the gui startup script to reset it back to 255 1 second after the gui is started. In several gui restarts and several reboots, the display brightness has remained set to the value set in the menu and through dbus-spy. Hopefully this will work on the 3B+ also.

pete10004x4, 2021-07-29:
When I try and run RpiDisplaySetup I get this:
`root@raspberrypi2:~# /data/RpiDisplaySetup/setup
ln: /data/RpiDisplaySetup/FileSets/v2.72/start-gui.sh.orig: File exists
ERROR: incomplete file set for v2.72 - can't continue`

Any ideas?

pete10004x4, 2021-07-29:
The contents of that directory is:
root@raspberrypi2:/data/RpiDisplaySetup/FileSets/v2.72# ls
INCOMPLETE                   start-gui.sh.NO_REPLACEMENT  start-gui.sh.orig

kwindrem, 2021-07-29:
Not sure why, but setup is attempting to create a file set for v2.72 and failing because a file it wants to link from another version already exists in that directory. That shouldn't be.
Try:
rm -r /data/RpiDisplaySetup/FileSets/v2.72

then run setup again. setup will attempt to create the file set again

There should be two sym links in that directory pointing back to the files in v2.60 file set.

pete10004x4, 2021-07-29:
No got this returned.
`root@raspberrypi2:~# /data/RpiDisplaySetup/setup
creating file set for v2.72
ERROR v2.72 start-gui.sh no replacement file
ERROR: incomplete file set for v2.72 - can't continue`

Let me do a clean install tomorrow on a clean sd card and see where we go.

Thanks for looking at it.

kwindrem, 2021-07-29:
My best guess is that /opt/victronenergy/gui/start-gui.sh was modified from the factory default.

If that was not done on purpose, you can restore it to stock by copying the file in the v2.60 file set to the .../gui directory.

If this was done on purpose, you won't be able to use my setup script. Instead, hand edit the file to add the nohup ... line near the end of the file.

My setup script compares the files in the working directories to copies with the .orig extension in the file set version directories. If it can't find a match, it can't create missing a file set which is the case for v2.72.

pete10004x4, 2021-07-30:
My apologies, something else I'm trying to do must have corrupted start-gui.sh, I ve done a clean install and run SetupHelper first and its installed as expected.

Just a thought would it be useful to rotate the display as part of the setup?

kwindrem, 2021-07-30:
I rotated mine to favor a viewing angle above the display.

closed 2021-07-30 by pete10004x4