kwindrem/RpiDisplaySetup #7: blank

Kraxxler82, 2021-12-09:
Hello,
thank you for your great work. it es very easy with your setup.

i have an waveshare 5"-Tochscreen running. 
i can change the brightness and i also can set the time for blanking.
the brightness works great, but the blanking only turns down the brightness, but the screen stays on.
what can i do?

kwindrem, 2021-12-09:
This depends on your display's capabilities.

My setup script insures /u-boot/config.txt includes the hdmi_blanking=1 line which should enable backlight control. But you could check the file to make sure the line is present and not commented out.

Many HDMI displays do not include the ability to turn off their backlight. I suppose this might also be true for displays connected via DSI or other means.

For those displays with no backlight control, the display simply dims and there's nothing you can do except to replace the display with one that does support backlight control.

Kraxxler82, 2021-12-11:
thank you for the information.

is it possible, to activate an black screensaver?

kwindrem, 2021-12-11:
when screen saver activates, the screen is supposed to be replaced with a black image, so you should see a dimly lit screen but all black. If that isn't happening, I'm not sure why.

Kraxxler82, 2021-12-12:
how can i activate the screen saver?

kwindrem, 2021-12-12:
Menu / Settings / Display & Language / Display off time