kwindrem/RpiDisplaySetup #10: LCD Rotate?

neoneddy, 2022-02-05:
Would be great to have a way to rotate the LCD from Remote Console.  Otherwise you can do this entirely without command line access! Great job!

morphias2004, 2022-03-03:
> Would be great to have a way to rotate the LCD from Remote Console. Otherwise you can do this entirely without command line access! Great job!

+1 for this enhancement request!  :)

For those unable to get it to work, I found that **lcd_rotate=2** does not work.

This is what worked for me:

Edit **/u-boot/config.txt**

Add the relevant section (or all) of the following parameters to the **[pi3+]** section and be sure to remove the hashes.

```
# For 0 degree rotation:
# (no additional entries - default setting)

# For 90 degree rotation (top to right):
#display_lcd_rotate=1
#dtoverlay=rpi-ft5406,touchscreen-swapped-x-y=1,touchscreen-inverted-x=1

# For 180 degree rotation (upside down)
#display_lcd_rotate=2
#dtoverlay=rpi-ft5406,touchscreen-inverted-x=1,touchscreen-inverted-y=1

# For 270 degree rotation (top to left)
#display_lcd_rotate=3
#dtoverlay=rpi-ft5406,touchscreen-swapped-x-y=1,touchscreen-inverted-y=1
```

kwindrem, 2022-03-03:
Interesting. On a Pi 4, all you need is the ice_rotate command. The dtoverlay for touch changes is not necessary.

RpiDisplay and VeCan are the only two packages that need command line interaction for setup. With the above, I'd also need to add the above lines based on a setup menu choice. I need to think about the best way to handle this. I really don't want to create new GUI menus for these configurations.

FYI, once set from the setup script command line menu, these settings will persist across package version updates, uninstall/reinstall or Venus OS updates. Also the settings backup/restore functionality can copy these from one system to another. Not what you are looking for but it is a step in the right direction.

neoneddy, 2022-03-03:
I'm not sure how much work it is for adding GUI options for LCD rotate is, but I've had a number of people ask on my Youtube Walk through on it.  They'd be thrilled.

jhedup, 2022-05-15:
On my Pi3+ with original 7inch LCD raspbian lcd_rotate=2 works. It does not on 2.84 or 2.85. Thanks for the work around morphias2004, I added it to /boot not /u-boot which did the trick,

PeterWagemans, 2023-02-07:
The ice_rotate command mentioned put me on the wrong leg. I think a lot of people who will use this excellent software (like me) are not very knowledgeable of the raspberry pi.
I installed everything using the GUI, and was so happy I got this superl cool display in my van now… :-) but when I tried to hang it to the wall, the holes for screwing in the enclosure were… with the small slit upside down. Which is not an execption, all enclosures turned out to be designed like that. So probably almost everyone has the problem sooner or later.

After a few hour (not kidding) of fiddling around, I found out how to start the setup through the command line, and use nano to change the config. And somehow got it working, and have the display upside down now.
I pray it keeps working, and having this in the GUI so I never ever have to do this again is my number 1 feature request :-)

@kwindrem thanks for all your work on this project!

kwindrem, 2023-02-08:
It is not practical to put config in the GUI since, well, you won't have the GUI until you've run the script. Also there are too many options to burden the GUI's menu system.

closed 2023-02-08 by kwindrem

neoneddy, 2023-07-01:
@kwindrem  I know this is an old closed issue.  Is is possible to have the config.txt as-edited just saved and restored on Venus OS update?  That would solve the issue for me.  

Thanks for the great package(s) continues to get better and better.

kwindrem, 2023-07-01:
config.txt is not changed by a Venus OS update.

neoneddy, 2023-07-01:
Ok, maybe we're chasing the wrong thing then.  Everytime I update venus OS, I have to rotate my screen.  If it was a one time thing, yeah, no big deal.  But every update I have to edit config.txt .  

I'm up to date on all your packages.  just updated to 3.0 vos