kwindrem/RpiDisplaySetup #12: Waveshare 7inch touch HDMI

jengstro2, 2022-04-12:
Needs following additional lines to u-boot\config.txt
```
hdmi_group=2
hdmi_drive=1
hdmi_ignore_edid=0xa5000080
hdmi_mode=87
hdmi_cvt=1024 600 60 6 0 0 0
```
Dunno, are all those required.
This is really odd display, it will need some time to warm up (quality....). With these settings. Give it time approx 5...10minutes, and the shadowing and glitching on display starts vanish.

jengstro2, 2022-04-13:
Touch "coordinates" are working with Rasp default image, don't know what's wrong with this display/venus-os combo. Have to say, there is some quality issues around these screen (by googling it).

jengstro2, 2022-04-14:
I noticed also other thing.. the display does not anymore sleep/shut-off. It did on 2.66 (now on 2.81)... I have defined "Display off time"

kwindrem, 2022-04-14:
My Waveshare HDMI display works properly. I'm running v2.84.

In RpiDisplaySetup's setup you need to select HDMI 0, not RPI. Then of course you don't get brightness control. Brightness only works with the RPI display or some other DSI displays.

kompassde, 2022-04-26:
Thanks for all this neat scripts @kwindrem 

I also use a the Waveshare 7 inch (Rev.2.1 / Type C) and the Firmware v2.85 of the VenusOS.
I installed the SetupHelper and the DisplaySetup. 

Somehow I am not able to get the Display to turn off.
With the Firmware 2.6 its working without problems.
Also with Firmware 2.73 its not working

Any suggestions to help me with my problem?

jengstro2, 2022-04-26:
> Somehow I am not able to get the Display to turn off. With the Firmware 2.6 its working without problems. Also with Firmware 2.73 its not working

Same here still for me also. I am not able to get it dimmed/sleep. I tried several configurations... no help.

kompassde, 2022-04-26:
I just reinstalled the RpiDisplaySetup and chose the HDMI 1... now the Display turned off after the set time.
There still seems to be tha backlight on but I think that is normal?

By the way, because it was mentioned before, my Waveshare also has a strange behavior. I needet to set the Frequency to 120! instead of 60 to get rid of the flickering. 
And still I have shadowing and glitching

jengstro2, 2022-04-26:
@kompassde can you share your current  u-boot\config.txt? I will try also.

kompassde, 2022-04-26:
Here are is my config.txt
but the key for me was to unistall the DisplaySetup and reinstall it with the HDMI Port 1
(with /data/RpiDisplaySetup/setup)

`[pi3]
device_tree=bcm2710-rpi-3-b.dtb
dtoverlay=miniuart-bt
core_freq=400
core_freq_min=400

hdmi_group=2
hdmi_mode=87
hdmi_cvt 1024 600 120 3 0 0 0
hdmi_drive=1`

kwindrem, 2022-05-17:
The unattended install will not choose the second HDMI port. You'll need to do that through the command line once. After that the option is stored and used for future reinstalls

Sysko75, 2024-03-30:
guys, how to activate da touch control on the wave share 7" HDMI.???? I try two days now and I can't figure out.