kwindrem/RpiDisplaySetup #13: canot install on Version 2.85

germai, 2022-04-20:
root@raspberrypi2:/data/RpiDisplaySetup# sh setup 
creating file set for v2.85
ERROR v2.85 start-gui.sh no replacement file
ERROR: incomplete file set for v2.85 - can't continue

I wont install it over PPackage manager or direck alway the sam problem.

its Pi 3b and the Pi 5"DSI LCD Display

kwindrem, 2022-04-20:
This error indicates the stock file the script is attempting to replace does not have a match. This usually means the file was modified from what is supplied in the Venus OS image.

The easiest way to recover is to switch to the backup firmware then reinstall the desired version from online update or via a USB stick.

germai, 2022-04-22:
Thanks now i can install ist.
but my problem is not fixt.
The Screenserver and the Dimmer do not work.
The dimmer is not in the menue and the Blech-Screen not work. 
Have you any Idee

Thanks

kwindrem, 2022-04-22:
Not all displays support dimming or blanking. It's also possible that specific display uses different /sys parameters. I have no way of knowing what they are, however if you discover what they are, you can manually enter them choosing the CUSTOM installation option.

germai, 2022-04-26:
Thanks. Ok. I Think de display ist the problem.
Have you e list of displays off display that support Slepping and Dimming
I look for a 3,5" 4.3" or 5" Display.
Thanks for Your support

kwindrem, 2022-04-26:
No list. I use the Raspberry Pi Official 7" Touchscreen. Others have had mixed success with other DSI displays.

No HDMI display supports dimming from the GUI, but some do support turing the backlight off for sleeping. Others will simply go to a black screen with backlight still on.

germai, 2022-04-27:
yes Raspberry Pi Official 7"  works now. 
Thanks

pr1m300, 2022-08-31:
Hey Guys,
 i have the same problem.

Iam using the Venus OS 2.90~26 on a Raspberry Pi 4 with the Official 7" Touchscreen.
When iam trying to install this tool i got always this message.

root@raspberrypi4:/data/RpiDisplaySetup# sh setup
--- starting setup script v2.8
creating file set for v2.90~26
ERROR v2.90~26 start-gui.sh no replacement file
ERROR: incomplete file set for v2.90~26 - can't continue

The touchscreen works so far, only I would like to have the brightness slider and the time-controlled deactivation of the display.

I have already tried different Venus OS versions that are available for the Raspberry Pi 4.
I hope someone can help me with this.
Thanks!

kwindrem, 2022-08-31:
RpiDisplaySetup v2.9 (the latest version) installed automatically on the on-line update I did this morning to v2.90~26. SetupHelper is v4.22.

The error you are reporting can occur if the file has been modified from the stock file supplied with the Venus OS image. You need to revert to a previous version then do a new update (not revert back to v2.90~26 in the stored backup). This will insure the file contents are what is expected and allow the setup script to run.

Each file to be replaced as part of a package is compared against version-dependent files included in the package. This tells the setup script which replacement file to use when installing the package. In the case of RpiDisplaySetup, only one file (start-gui.sh is replaced and there are only two variants: one for v2.66 and one for v2.80. The copy of that file included in Venus OS will match one of these variants (probably the v2.80).

pr1m300, 2022-09-01:
I got it now. 
I tried Venus OS v2.80 first and it worked. But I couldn't update to Venus OS 2.90 from there, so I flashed Venus OS 2.90~26 again to an SD card. This time I installed the SetupHelper first and then I was able to install RpiDisplaySetup without any problems.

Now everything works great! Thanks.

closed 2022-09-01 by kwindrem