kwindrem/RpiDisplaySetup #17: Waveshare 4.3 ; No Display-off

claib, 2022-12-20:
Hi there
works all like a charm : thx so much for the great work !
One thing is missing:
The screen does not go dark !
Even the brigtness slider works perfect, so the solution can't be so far away ;-)
I can set 1 minute Display off time, but display is not switching off

any advice ?
thx

kwindrem, 2022-12-20:
You'll need to poke around in /sys/class to find the proper screen power device file. Sorry, I can't help with a specific display since I don't have access to the hardware.

sarah1331sarah, 2023-10-08:
hi i also have a 4.3 touch screen dis it dims but doesnt turn off it will dim on the "power off" timer can you please elabrote on the /sys/class what are we looking to do, access, change thank you for your time

kwindrem, 2023-10-08:
While some HDMI displays will turn off their backlight when receiving no signal from HDMI but many do not. If the display does not turn off it's backlight, then you'll see a faint glow from the black screen.

There really is not much that can be done from the Venus OS side to change this behavior.

sarah1331sarah, 2023-10-08:
Thanks for a response does it matter I am not using hdmi the screen would be good to go blank but it just dims to the lowest setting but still shows the full screen if that makes sense

kwindrem, 2023-10-08:
DSI displays are typically better behaved and with more features (like adjustable brightness), but there are just too many displays out there all with different behavior. What a display does with the blanking and dimming information is ultimately up to the display.

sarah1331sarah, 2023-10-08:
I understand now the dim/ turn off is sent to the display but what it does with the command is programmed by the screen manufacturer (or not programmed) hence the result 

thanks again for your responses