kwindrem/RpiDisplaySetup #18: Cannot get touch display to work

NickDolfin, 2022-12-20:
I use a virgin Raspi pi 3b+ with a 5"waveshare lcd (https://www.waveshare.com/wiki/5inch_HDMI_LCD) and installed venus-image-large-raspberrypi2-202210162113031-v2.92.rootfs.wic.gz

Install of setuphelper (https://github.com/kwindrem/SetupHelper) works.
Then I installed https://github.com/kwindrem/RpiDisplaySetup
(fiddled around in the victron menu's to get it activated)

in a root terminal I did the following:
cd /data/RpiDisplaySetup
bash setup
H
C
800
480
cr
5 (16:10)
n
n
y

Then the system reboots, but there is no touch calibration.
Display looks nice tough.

Any idea what I am doing wrong?

kwindrem, 2022-12-21:
The touchscreen calibration app is not installed by default. the RpiDisplaySetup script should attempt to install it but if internet connection isn't present that would fail. And if it isn't present, the script won't attempt to run the calibration.

The log may shed some light on things:

tail -50 /data/log/SetupHelper | tai64nlocal

Sysko75, 2024-03-30:
hi, i do exact the same now, but after first rebooting I get from screen a 4 color square image, nothing from the GUI what going wrong?

kwindrem, 2024-03-30:
Check the log (as mentioned above) for any indication of setup problems.

Sysko75, 2024-03-30:
root@192.168.1.22's password: 
Last login: Sat Mar 30 15:20:35 2024
root@raspberrypi4:~# tail -50 /data/log/SetupHelper | tai64nlocal
2024-03-30 15:09:14.513072500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:09:15.879347500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:09:16.349872500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:09:17.159607500 RpiDisplaySetup/setup: restarting generator service
2024-03-30 15:09:17.219702500 RpiDisplaySetup/setup: complete - no errors
2024-03-30 15:08:41.744938500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:08:41.762276500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:08:41.770428500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:08:41.759062500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:08:41.777983500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:08:41.785933500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:14:28.737587500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:14:28.760749500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:14:28.768828500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:14:28.738609500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:14:28.756446500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:14:28.764563500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:18:42.979916500 RpiDisplaySetup/setup: --- starting setup script v3.8 action NONE
2024-03-30 15:18:43.180695500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:19:33.689727500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:19:33.798564500 RpiDisplaySetup/setup: script will run again at startup
2024-03-30 15:19:33.814945500 RpiDisplaySetup/setup: completed - reboot needed
2024-03-30 15:19:43.764940500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:19:43.782448500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:19:43.790627500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:20:16.372355500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:20:17.877428500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:20:18.307996500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:20:18.558420500 RpiDisplaySetup/setup: running touchscreen calibration
2024-03-30 15:20:18.802446500 RpiDisplaySetup/setup: activating local GUI
2024-03-30 15:20:18.886926500 RpiDisplaySetup/setup: script will run again at startup
2024-03-30 15:20:18.935564500 RpiDisplaySetup/setup: completed - reboot needed
2024-03-30 15:20:29.743234500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:20:29.760826500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:20:29.768815500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:21:02.648908500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:21:03.678524500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:21:04.258443500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:21:04.928395500 RpiDisplaySetup/setup: restarting generator service
2024-03-30 15:21:05.008012500 RpiDisplaySetup/setup: complete - no errors
2024-03-30 15:23:42.244605500 ShutdownMonitor/setup: --- starting setup script v3.1 action CHECK
2024-03-30 15:23:42.269010500 ShutdownMonitor/setup: creating package options directory /data/setupOptions/ShutdownMonitor
2024-03-30 15:23:47.806810500 ShutdownMonitor/setup: --- starting setup script v3.1 action INSTALL
2024-03-30 15:23:48.044340500 ShutdownMonitor/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:23:48.141427500 ShutdownMonitor/setup: installing files
2024-03-30 15:23:48.255517500 ShutdownMonitor/setup: installing services
2024-03-30 15:23:48.265258500 ShutdownMonitor/setup: installing ShutdownMonitor service
2024-03-30 15:23:51.567302500 ShutdownMonitor/setup: service ShutdownMonitor running
2024-03-30 15:23:51.599120500 ShutdownMonitor/setup: restarting generator service
2024-03-30 15:23:51.621233500 ShutdownMonitor/setup: completed - GUI restart needed
root@raspberrypi4:~# D

Sysko75, 2024-03-30:
`root@192.168.1.22's password: 
Last login: Sat Mar 30 15:20:35 2024
root@raspberrypi4:~# tail -50 /data/log/SetupHelper | tai64nlocal
2024-03-30 15:09:14.513072500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:09:15.879347500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:09:16.349872500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:09:17.159607500 RpiDisplaySetup/setup: restarting generator service
2024-03-30 15:09:17.219702500 RpiDisplaySetup/setup: complete - no errors
2024-03-30 15:08:41.744938500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:08:41.762276500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:08:41.770428500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:08:41.759062500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:08:41.777983500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:08:41.785933500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:14:28.737587500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:14:28.760749500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:14:28.768828500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:14:28.738609500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:14:28.756446500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:14:28.764563500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:18:42.979916500 RpiDisplaySetup/setup: --- starting setup script v3.8 action NONE
2024-03-30 15:18:43.180695500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:19:33.689727500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:19:33.798564500 RpiDisplaySetup/setup: script will run again at startup
2024-03-30 15:19:33.814945500 RpiDisplaySetup/setup: completed - reboot needed
2024-03-30 15:19:43.764940500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:19:43.782448500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:19:43.790627500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:20:16.372355500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:20:17.877428500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:20:18.307996500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:20:18.558420500 RpiDisplaySetup/setup: running touchscreen calibration
2024-03-30 15:20:18.802446500 RpiDisplaySetup/setup: activating local GUI
2024-03-30 15:20:18.886926500 RpiDisplaySetup/setup: script will run again at startup
2024-03-30 15:20:18.935564500 RpiDisplaySetup/setup: completed - reboot needed
2024-03-30 15:20:29.743234500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:20:29.760826500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:20:29.768815500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:21:02.648908500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:21:03.678524500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:21:04.258443500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:21:04.928395500 RpiDisplaySetup/setup: restarting generator service
2024-03-30 15:21:05.008012500 RpiDisplaySetup/setup: complete - no errors
2024-03-30 15:23:42.244605500 ShutdownMonitor/setup: --- starting setup script v3.1 action CHECK
2024-03-30 15:23:42.269010500 ShutdownMonitor/setup: creating package options directory /data/setupOptions/ShutdownMonitor
2024-03-30 15:23:47.806810500 ShutdownMonitor/setup: --- starting setup script v3.1 action INSTALL
2024-03-30 15:23:48.044340500 ShutdownMonitor/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:23:48.141427500 ShutdownMonitor/setup: installing files
2024-03-30 15:23:48.255517500 ShutdownMonitor/setup: installing services
2024-03-30 15:23:48.265258500 ShutdownMonitor/setup: installing ShutdownMonitor service
2024-03-30 15:23:51.567302500 ShutdownMonitor/setup: service ShutdownMonitor running
2024-03-30 15:23:51.599120500 ShutdownMonitor/setup: restarting generator service
2024-03-30 15:23:51.621233500 ShutdownMonitor/setup: completed - GUI restart needed
root@raspberrypi4:~# D

`

Sysko75, 2024-03-30:
root@192.168.X.XX's password: 
Last login: Sat Mar 30 15:20:35 2024
root@raspberrypi4:~# tail -50 /data/log/SetupHelper | tai64nlocal
2024-03-30 15:09:14.513072500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:09:15.879347500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:09:16.349872500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:09:17.159607500 RpiDisplaySetup/setup: restarting generator service
2024-03-30 15:09:17.219702500 RpiDisplaySetup/setup: complete - no errors
2024-03-30 15:08:41.744938500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:08:41.762276500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:08:41.770428500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:08:41.759062500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:08:41.777983500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:08:41.785933500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:14:28.737587500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:14:28.760749500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:14:28.768828500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:14:28.738609500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:14:28.756446500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:14:28.764563500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:18:42.979916500 RpiDisplaySetup/setup: --- starting setup script v3.8 action NONE
2024-03-30 15:18:43.180695500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:19:33.689727500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:19:33.798564500 RpiDisplaySetup/setup: script will run again at startup
2024-03-30 15:19:33.814945500 RpiDisplaySetup/setup: completed - reboot needed
2024-03-30 15:19:43.764940500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:19:43.782448500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:19:43.790627500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:20:16.372355500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:20:17.877428500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:20:18.307996500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:20:18.558420500 RpiDisplaySetup/setup: running touchscreen calibration
2024-03-30 15:20:18.802446500 RpiDisplaySetup/setup: activating local GUI
2024-03-30 15:20:18.886926500 RpiDisplaySetup/setup: script will run again at startup
2024-03-30 15:20:18.935564500 RpiDisplaySetup/setup: completed - reboot needed
2024-03-30 15:20:29.743234500 SetupHelper/reinstallMods: reinstallMods starting
2024-03-30 15:20:29.760826500 SetupHelper/reinstallMods: SetupHelper already installed
2024-03-30 15:20:29.768815500 SetupHelper/reinstallMods: PackageManager will reinstall remaining packages
2024-03-30 15:21:02.648908500 RpiDisplaySetup/setup: --- starting setup script v3.8 action INSTALL
2024-03-30 15:21:03.678524500 RpiDisplaySetup/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:21:04.258443500 RpiDisplaySetup/setup: ++ Installing display config to /u-boot/config.txt for display: HDMI
2024-03-30 15:21:04.928395500 RpiDisplaySetup/setup: restarting generator service
2024-03-30 15:21:05.008012500 RpiDisplaySetup/setup: complete - no errors
2024-03-30 15:23:42.244605500 ShutdownMonitor/setup: --- starting setup script v3.1 action CHECK
2024-03-30 15:23:42.269010500 ShutdownMonitor/setup: creating package options directory /data/setupOptions/ShutdownMonitor
2024-03-30 15:23:47.806810500 ShutdownMonitor/setup: --- starting setup script v3.1 action INSTALL
2024-03-30 15:23:48.044340500 ShutdownMonitor/setup: patching VisibleItemModel to VisualItemModel in all .qml replacements
2024-03-30 15:23:48.141427500 ShutdownMonitor/setup: installing files
2024-03-30 15:23:48.255517500 ShutdownMonitor/setup: installing services
2024-03-30 15:23:48.265258500 ShutdownMonitor/setup: installing ShutdownMonitor service
2024-03-30 15:23:51.567302500 ShutdownMonitor/setup: service ShutdownMonitor running
2024-03-30 15:23:51.599120500 ShutdownMonitor/setup: restarting generator service
2024-03-30 15:23:51.621233500 ShutdownMonitor/setup: completed - GUI restart needed
root@raspberrypi4:~# D

Sysko75, 2024-03-30:
sorry I don't understand why can not paste with normal font SORRY about that

kwindrem, 2024-03-30:
I don't see any errors in the log so I suspect that the settings you provided when running the setup script are not compatible with your display. It's not really possible for me to provide support for specific displays since variations are endless.

Seeing the colored square during boot is actually a good sign.

It's possible that something has set headless mode which disables the local GUI. Try:

```
rm -f /etc/venus/headless
reboot

```

Look at the specs for your display and verify that the parameters you entered for the display are correct. RpiDisplaySetup modifies /u-boot/config.txt so you can check that too. Make sure the hdmi group and and the mode number are correct for your display. Also make sure the display supports the resolution you specified.

Getting an HDMI display working can be a challenge and often requires experimentation. It may be worth running Raspberry PI OS to see if the display works with that before trying to get Venus OS to accept it.

Hope this helps

Sysko75, 2024-03-30:
👍🔝💪 all fine now! Thanks

manumeadow, 2025-01-18:
Hi there,
I'm also interested, which solution @Sysko75 found.
I'm using an old 13"ish touchscreen, image is shown correctly, there's also siome touch functionality, but it's totally off for it's not calibrated.
If I start calibration via /data/RpiDisplaySetup/setup there are no instructions nor squares to touch on to calibrate after reboot. There's only one small green square in the middle which doesn't react to touch gestures. Venus OS is starting right through.
I can't find any logs under /data/log/SetupHelper . Also it seems to me as if /etc/venus/headless is not used anymore for disabling local gui.
I'm thankful for any help, I think I'm missing small things or there has been some changes in Venus OS.
Regards Manuel

kwindrem, 2025-01-18:
It appears Victron has changed the GUI startup mechanism as part of v3.60 beta. The GUI always starts and as you guessed /etc/venus/headless no longer is used !!!!! I'm going to have to redo the RpiDisplaySetup setup script to allow calibration.

For now you can do the touch screen calibration manually from the command line:

svc -d /service/gui
ts_calibrate
reboot

This assumes ts_calibrate has already been installed so I suggest you run /data/RpiDisplaySetup/setup and select install then calibrate only which will fail as you have reported here.

I'll get a new version of RpiDisplaySetup out today.

kwindrem, 2025-01-19:
v3.15 fixes the issue with calibration not working with v3.60...

An added benefit of the new GUI code (Classic too) is that quitting the GUI releases the frame buffer allowing ts_calibrate to run without a reboot. So it's much faster.

closed 2025-01-19 by kwindrem