kwindrem/RpiDisplaySetup #19: High CPU load with 7" HDMI display

fibrinogen, 2023-01-04:
Hi! I installed a 72 HDMI display on a raspberry pi 4. The display and touch functionality works. But when the display is connected the raspberry gets very slow and it is not possible to connect via ssh or web. The display is working, but nearly unusable, because it reacts very slow or not at all.
I used the console install and choose:
(i)
(H)
(A)
(n)
(n)
(y)
After reboot the touch screen calibration starts.
After next reboot the display is working, but everything is incredibly slow or not reacting at all.

kwindrem, 2023-01-04:
Not sure what to suggest as I've not seen this. If you can disconnect the display, reboot then ssh in, maybe you can get an idea of what's loading down the system. 'top' may be of some use. After you are logged in, try connecting the display again.

fibrinogen, 2023-01-05:
Thanks for your reply! 
Id did what you mentioned, but in the moment when connecting the display the system nearly freezes and top does not refresh any more. But then I used the manual configuration option (C) with:
hdmi_group=2
hdmi_mode=87
display_rotate=0
hdmi_cvt 1024 600 60 6 0 0 0
and after a restart it worked flawlessly! In case others are searching for information: the display type is JRP7002

closed 2023-01-05 by fibrinogen