kwindrem/RpiDisplaySetup #20: Display blank not working

AirTobe91, 2023-01-27:
Hello,

I have setup a Raspberry Pi 4 with an 5,5" Waveshare Amoled FHD Display attached via HDMI.
I'm using VenusOS 2.9.2.
I got everything working, just the blanking isn't working.

Currently I set the timeout to 30sec, but nothing is working.
I've checked the following:

in /u-boot/config.txt:
hdmi_blank=1 

rpi-backlight-overlay.dtb and rpi-display-overlay.dtb available in /u-boot/overlays

/etc/venus/blank_display_device available and points to /sys/class/graphics/fb0/blank which is also available.

Do you maybe have an idea, what else I could check to get it working?
Otherwise, the AMOLED display is self destroying after some hours, because of burn-in :-(

Thank you!

kwindrem, 2023-01-27:
check to see if the device actually blanks the display:

echo 1 > /sys/class/graphics/fb0/blank

echo 0 > /sys/class/graphics/fb0/blank

If that doesn't work, try looking around in /sys/class for other devices that might blank the display. If you find one you can put the new device in 

/etc/venus/blank_display_device

AirTobe91, 2023-01-28:
the "echo 1 > /sys/class/graphics/fb0/blank" says "bash: echo: write error: Invalid argument"

It's interesting...
the /sys/class/graphics/fb0/blank points to /sys/devices/platform/3dbd7000.framebuffer/graphics/fb0/blank
But I'm somehow not able write a 0 or 1 to this file.. 
If I'm opening the file with "vi" it's saying, it can't read the file.
The read/write rights to the file are ok...

AirTobe91, 2023-01-28:
I just checked and I'm not able to modify, create or delete any file under /sys/ even if I set recursively write rights for root...
Is this normal?

chmod -R u+w /sys
cd /sys
touch test.txt
-> touch: test.txt: Permission denied

So I can imagine, it's no possible for any process to write a 0/1 into /sys/class/graphics/fb0/blank file, and that's why there is no blanking happening?!

kwindrem, 2023-01-28:
If you can't read or write the file it's likely that fb0 isn't the device being used for your display.

See if there is anything in /sys/class/backlight and if so follow that tree.

I'm using the Official Raspberry PI 7" touchscreen and it shows up under /sys/class/backlight/rpi_display. The blank file is bl_power.

/sys is a special directory tree and you should not be able to create files. I wouldn't try to change permissions.

AirTobe91, 2023-01-28:
Ah, that was the correct hint!
It's not fb0, instead it's fb1. If I'm doing an "echo 1 > /sys/class/graphics/fb1/blank" then it's turning off.

If adapted the /etc/venus/blank_display_device for fb1 and now it's working well :-)
Thanks for your fast help!

closed 2023-01-28 by AirTobe91