kwindrem/RpiDisplaySetup #22: Display turns short on, off and on after awaking

AirTobe91, 2023-02-26:
Hello,

I sometimes have a strange behavior, when awaking my Waveshare 5" Amoled Display from blanking.
Sometimes when I touch the blanked screen, the screen turns on, 1 sec later off, 1sec later again on and then it stays on.
It's interesting, that it's not every time, I would say every second try...

Did you already experience such a problem? Could it be due to the display itself, or maybe something with the Raspberry?

Thank you!

kwindrem, 2023-02-26:
I have no idea what could be causing this. It's either something in the display itself or in Venus Os. Nothing in RpiDisplaySetup could be causing it since that script just makes some connections then lets logic deep down in Venus OS manage the display.

AirTobe91, 2023-02-26:
Ok, thank you! I consulted the display manufacturer and will see if there is maybe some additional configuration needed.

closed 2023-02-26 by AirTobe91