kwindrem/RpiDisplaySetup #24: Running Setup disables CAN Hat

badfrogg, 2023-09-10:
After running: /data/RpiDisplaySetup/setup , I loose the CAN Hat (VeCanSetup).

I have to reinstall it every time.

kwindrem, 2023-09-10:
Need more information:

Which CAN hat?

How are you installing it when it does work?

badfrogg, 2023-09-10:
I am using the waveshare 2-ch isolated (Non-FD)

I am installing it through the GUI Package manager interface.

I should clarify, I am nopt loosing the installation of VeCanSetup. The can hat just stops working and reinstalling VeCanSetup brings it back.

kwindrem, 2023-09-10:
You can not run VeCanSetup from PackageManager as it requires console interaction to configure the interfaces.

You must run /data/VeCanSetup/setup from a command line prompt either via ssh or a direct console connection.

Once you have added and configured ports via the command line., these settings will be restored when updating Venus OS (aka firmware update).

closed 2023-09-10 by kwindrem

badfrogg, 2023-09-10:
I did run VeCanSetup from the command line the first time I installed it. That worked fine. I was just reporting to you that running RpiDisplaySetup makes the CAN bus stop working.

It is not a problem for me becaus my screen is working now and I don't have a reason to run RpiDisplaySetup anymore. I just though you might like to know about the behavior. 

Thanks for contribution by the way!

kwindrem, 2023-09-11:
fixed in v3.4

Thanks for the report, sorry for my misunderstanding.