kwindrem/RpiDisplaySetup #27: Help with 3.5" TFT screen

jacklawry, 2023-10-10:
I have this screen (on a RPi2) it was working fine when running normal RPi but now I have moved this RPi to use on Victron it just shows as a white screen in Venus. 

https://thepihut.com/products/abs-case-with-3-5-tft-touchscreen-for-raspberry-pi-4-480x320

Any help much appreciated.

kwindrem, 2023-10-10:
A white screen usually indicates a corrupt GUI file.

Try refreshing the SD card with a fresh Venus OS.

jacklawry, 2023-10-10:
Ok, I am just flashing a new card now with `venus-image-large-raspberrypi2.wic.gz` but this screen has never worked with Venus it has always just shows white even with a fresh install and power on... Not been able to follow the normal [install steps](https://thepihut.com/products/abs-case-with-3-5-tft-touchscreen-for-raspberry-pi-4-480x320) as no git (and no apt to install git)

kwindrem, 2023-10-10:
This display may need a driver that is not compatible with Venus OS.

Venus OS is scaled back from what you may be used to on a Raspberry PI running raspberry pi os. 

You may be able to use wget in place of git clone, or do the git on another machine then rsync or rcp the folder to the PI.

apt is not installed but opkg is.

jacklawry, 2023-10-10:
Unfortunately it doesn't work as it seems to require `apt-get` to actually work even once it has the drivers. 

```
root@raspberrypi2:~/LCD-show# sudo ./LCD35-show
grep: /boot/config.txt: No such file or directory
grep: invalid option -- 'P'
BusyBox v1.31.1 (2023-10-09 06:00:11 UTC) multi-call binary.

Usage: grep [-HhnlLoqvsriwFE] [-m N] [-A/B/C N] PATTERN/-e PATTERN.../-f FILE [FILE]...
cp: can't stat '/boot/config.txt': No such file or directory
cp: can't stat '/boot/cmdline.txt': No such file or directory
cp: can't create '/usr/share/X11/xorg.conf.d/99-fbturbo.conf': No such file or directory
cp: can't stat '/etc/rc.local': No such file or directory
cp: can't stat '/etc/modules': No such file or directory
cp: can't create '/boot/overlays/': Is a directory
cp: can't create '/boot/overlays/tft35a.dtbo': No such file or directory
./system_config.sh: line 3: lsb_release: command not found
cat: can't open '/etc/debian_version': No such file or directory
cp: can't create '/usr/share/X11/xorg.conf.d/99-fbturbo.conf': No such file or directory
sudo: apt-get: command not found
sudo: apt-get: command not found
install cmake error!!!!
2023
need to update touch configuration
sudo: dpkg: command not found
cp: can't stat '/usr/share/X11/xorg.conf.d/10-evdev.conf': No such file or directory
reboot now
```

Thanks for your help but I guess I will just  have to run it headless for now.