kwindrem/RpiDisplaySetup #28: Blanking waveshare 7

PatrickB100, 2023-11-18:
Goodmorning,

is there a status file which indicates if the blanking is active or not? The waveshare 7 needs to be switched of by an io which i can do with node red. 

Thanks,
Patrick

kwindrem, 2023-11-18:
If you are talking about turning the screen off and on (the screen saver function), then you can probably read the file pointed to by /etc/venus/blank_display_device.

RpiDisplaySetup allows this to be configured manually from the command line if you know what /sys/class... is needed for screen saver control.

mcgyver78, 2023-12-09:
i´ve found a solution 

blanking the display an switching  off the background light 

with this 
https://community.victronenergy.com/questions/146206/cheap-35-tft-touchscreen-backlight-raspberry-pi-mo.html

read the comments

PatrickGoettsch, 2024-01-04:
> i´ve found a solution
> 
> blanking the display an switching off the background light
> 
> with this https://community.victronenergy.com/questions/146206/cheap-35-tft-touchscreen-backlight-raspberry-pi-mo.html
> 
> read the comments

Hi, I tried your solution but still my screen is not blanking. I am running Victron OS 3.13. Have you any further idea?

mcgyver78, 2024-01-04:
do you measure the GPIO18 / PIN2  ? 
make sure the pin is switching with the blanking time you have set. 

backlightCtrl /dev/input/touchscreen0 45