kwindrem/RpiDisplaySetup #30: Raspberry Pi 7inch Display Rotation

FunkFamily, 2023-12-18:
Using the raspberry pi's 7 inch touchscreen with the lcd ribbon, the display works after installing "RpiDisplaySetup" but you have to edit config.txt in the u-boot directory with "lcd_rotate=2" to rotate in the proper orientation so that it works with all display housings.  I'll provide a picture of what it looks like after editing the config.txt file
[image]

kwindrem, 2023-12-18:
When you run RpiDisplaySetup's setup script, you are asked if you want to rotate the display. If so, the script adds that line to config.txt.

FunkFamily, 2023-12-18:
I have never been asked after install. Is this something you have to initiate in a terminal

kwindrem, 2023-12-18:
While you can auto-install RpiDisplaySetup from PackageManager, doing so uses defaults: the official RPI touchscreen with no rotation, default backlight and dimming.

if you need anything else, you need to run /data/RpiDisplaySetup/setup from the terminal or ssh. You are then prompted for your specific options.

closed 2023-12-18 by kwindrem