kwindrem/RpiDisplaySetup #32: RpiDisplaySetup V3.7 must be run from the command line

58wolf, 2023-12-30:
Hi

I have  install Venus Large, Helper, Packagemanager and some packages on a Raspberry pi4. 

But RPiDIsplaySetup cannot be installed. Only from the command line.

I´m new in linux and need some info. 
I know to go via ssh to the rasp. 
where is the download package?
which command I need to install the package and if necessary with the Directory and filename and option for the command

Greets

58wolf, 2023-12-30:
I install the PI into a 10" Touch 

[image]

kwindrem, 2023-12-30:
Looks like you got things installed.

If not the command to get to the setup script is

/data/RpiDisplaySetup/setup

Then follow the prompts

The ReadMe file in the package directory (/data/RpiDisplaySetup) or on GitHub will explain the choices more.

58wolf, 2023-12-30:
Thks

But which choice I use? 

Rasperry R , HDMI (H) or skip to config?

Resolution of the touchscreen is 1280x800

kwindrem, 2023-12-30:
I can't answer your questions because I do not know what kind of display you are using. If you are plugging into an HDMI port on the PI, try that selection.

58wolf, 2023-12-30:
THks for help.
usd HDMI and than automaticly configure.

[image]

58wolf, 2023-12-30:
[image]

closed 2023-12-30 by kwindrem