kwindrem/RpiDisplaySetup #33: Cannot get 7" Touch screen working for touch

Zeoic, 2024-02-04:
Hello,

I am just setting this up for the first time, and I was able to get the package installed, ran the setup, got it to do touch calibration which it seems to have completed, but it is still not working for touch. The local GUI loads up just fine, but touch does nothing. I was able to confirm touch works via a windows PC, so it shouldn't be faulty hardware.

I ran "tail -50 /data/log/SetupHelper | tai64nlocal" and can see in the log that it did indeed run touch calibration, but nothing really happened after that. The rest of the log looks like the normal boot process.

Any suggestions?
Thanks!

kwindrem, 2024-02-04:
The calibration requires you to touch the screen in 5 places, then the PI should reboot. The GUI should then come up.

Is that what you are seeing?

What you describe in the log seems correct.

The command to do the calibration manually is ts_calibrate not ts_setup.

Zeoic, 2024-02-04:
Ts_calibrate is what I am running. When i run it, "ts_setup: No such file or directory" is what gets outputted. Nothing happens on the screen for touch calibration, just the single line of output

kwindrem, 2024-02-04:
sounds like ts_calibrate didn't install properly. Try that manually

	opkg update
	opkg install tslib-calibrate

 then try to run ts_calibrate from the command line

(you need to set the pi to headless for the calibration: touch /etc/venus/headless; reboot)

Zeoic, 2024-02-04:
I had tried that already as well, after seeing you mention it on the victron forums. I just get the output saying no packages were installed, which I presume means it already is installed. Maybe its worth try to uninstall it then reinstall?

kwindrem, 2024-02-04:
That would be my next step.

Zeoic, 2024-02-04:
Alright I will be back near the pi in a bit and will give that a try

Zeoic, 2024-02-04:
A reinstall didn't help. I have been looking into things a bit more, and it seems like the touch screen is just not showing up as a device at all. I tried plugging in other devices like keyboards and those register just fine, but nothing happens with the touch screen.. 
Do some panels need special drivers by any chance?

kwindrem, 2024-02-04:
The need for drivers is possible. Check the requirements for the device. It may not be possible to use this display with Venus OS if so.

Zeoic, 2024-02-04:
I did some testing on other linux distros, on the pi and on my desktop, and I can't get touch working anywhere (other than windows). I am just going to return this one. 

Would you happen to have a recommendation for a brand or specific 7" - 10" screen that you would suggest? I should probably get compatible hardware this time lol

kwindrem, 2024-02-04:
I have used both of these:

https://www.amazon.com/gp/product/B0153R2A9I/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

https://www.amazon.com/gp/product/B077PLVZCX/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

The first one has no case but will flush mount to a wall with a little tinkering. It also supports dimming which HDMI displays do not.

Zeoic, 2024-02-04:
That first one looks perfect, thanks! I will be printing a face plate for a diy battery box that this will go in, so not having a case works great. Thanks a ton for all the help!

Zeoic, 2024-02-06:
Working great with the first-party raspberry pi screen. Thanks again!

closed 2024-02-06 by Zeoic