kwindrem/RpiDisplaySetup #35: no fileset for v3.22

miramar-labs, 2024-03-17:
trying to install RpiDisplaySetup v3.7 using latest PackageManager on my Rpi4 and getting this error....
any idea what I'm doing wrong?

kwindrem, 2024-03-17:
while v3.7 has no specific file set for v3.22 SetupHelper should be able to create one. You might see a no file sets for a few seconds (1 second per stored package) while the missing file set is build but it should disappear and Install button should enable. If the file set could not be created the message should turn to incomplete file set and Install should not enable.

Make sure you have at least SetupHelper v7.0 installed. v7.1 was released earlier today.

In previous versions of SetupHelper, no file set meant two things: either the file set was missing or that it could not be created during in install attempt. If the latter, this probably meant that something else modified /opt/victronenergy/gui/start-gui.sh.

I just released RpiDisplaySetup v3.8 that will contain a file set for v3.22. 

Starting with SetupHelper v7.0, the original package modifying a file is stored on the system and this will be reported as a package conflict. BUT if the modification is happening manually or via some other mechanism that does not use SetupHelper, then installing RpiDisplaySetup will overwrite those modifications! 

Note that both v3.7 and v3.8 install without errors on Venus OS v3.22 on my test system.

If you continue to have difficulties and have not intentionally modified start-gui.sh then I suggest you reinstall the firmware. You can do that by switching to the backup firmware then using the Online update to install the current version again.

Contact me here if you still can't resolve the issue.

miramar-labs, 2024-03-17:
Getting the same thing .. so I must be doing something silly (I'm new with Venus OS, but not Linux). See attached pic.
[image]

So all I did was get a new RPi 4, and flashed the SD card with:
https://updates.victronenergy.com/feeds/venus/release/images/raspberrypi4/venus-image-raspberrypi4.wic.gz  

2024-02-26 15:22 | 121M

(not sure if I was supposed to use the 'large' version?)

Set up SSH login .. wifi and removed the /etc/venus/headless file ... then installed the SetupHelper... and then used that to download/install RpiDisplaySetup.
I haven't modified start-gui.sh or anything like that...

kwindrem, 2024-03-17:
The must be run from command line indicates a package (RpiDisplaySetup in this case) needs information that can only be supplied by running /data/RpiDisplaySetup/setup from the command line (ssh or local terminal). Once this has been done once, PackageManager can then install and uninstall it without command line interaction.

The large version is not required for SetupHelper, GuiMods or RpiDisplaySetup as well as any of MY other packages. There may be packages authored by others that do require large so check their documentation.

miramar-labs, 2024-03-17:
ok running that setup script did the trick! Now all your packages show as installed and my screen is calibrated!
Thanks for the help and thankyou for your work on this.

closed 2024-03-17 by miramar-labs