kwindrem/RpiDisplaySetup #36: installation issue

Sysko75, 2024-03-27:
on my new rasp4 I have installed Venus 2.93 today, installed GUImods 10.17, setup helper 7.11 and all work fine, now I try to connect my HDMI 7" screen, I try to install rpiDisplaySetup but before finish I get a message that I need proced by command line... and on the screen I can see only the root... 
what I have to do ???? please help thanks

kwindrem, 2024-03-27:
RpiDisplaySetup requires that you ssh in to the PI and run the setup script from there. There are many prompts to provide information about your system and these can not be asked through the GUI.

Sysko75, 2024-03-27:
Ok thanks, but were can I find a tutorial to follow or instruction for resolve

closed 2024-03-27 by kwindrem