kwindrem/RpiDisplaySetup #38: Installation fails

MBj1703, 2024-07-11:
Hi,

I try to install the actual version on a RaspPi 3B with Venus OS large v3.34
The SetupHelper is installed and working, nothing else.
I got the following error: RiDisplaySetup install failed - setup must be run from the command line

[image]

kwindrem, 2024-07-11:
This is not really a failure. RpiDisplaySetup needs input from the installer (person) in order to make the correct configuration changes. You need ssh access to your Raspberry PI in order to perform the install.

closed 2024-07-11 by kwindrem

MBj1703, 2024-07-11:
??

Installation:

RpiDisplaySetup requires that SetupHelper is installed first.

The easiest way to install RpiDisplaySetup is to do a "blind install" of SetupHelper
	and then add the RpiDisplaySetup package via the PackageManager menus.

Refer to the SetupHelper ReadMe for instructions and for other install options:
	https://github.com/kwindrem/SetupHelper