kwindrem/RpiDisplaySetup #41: Support for Official Raspberry Pi 7″ Touch Display 2 ?

LykkeGit, 2024-11-22:
I'm trying to get the RpiDisplay to run on a RPi4 with the new Raspberry Pi 7" Touch Display 2. SetupHelper and RPiDisplaySetup installs without issues, but I can't seem to get the display to work.  

Is Touch Display 2 supported?

kwindrem, 2024-11-22:
I was not aware a new display was released. I also am not sure why it doesn't work with Venus OS. It's possible a new device tree overlay or other driver is needed for it that is not included in the Venus OS image.

LykkeGit, 2024-11-23:
Ok, thank you, that explains. I'll try and see if I can swap it with the old model from some other project. PS: Thanks for all your fantastic tools to support Venus on RPi's - much appreciated.

MalcolmSidney, 2025-01-12:
@LykkeGit did you have any luck with the older display? I am also stuck getting the newer version running.

LykkeGit, 2025-01-12:
@MalcolmSidney , I swapped the new display with a Chinese 7” display from another project. The Aliexpress display works without issues.
https://www.aliexpress.com/item/1005005641383181.html

LykkeGit, 2026-02-01:
> I swapped to the older 7 inch screen to get mine working.

Same here - used an old screen. Maybe it has been updated by now?

kwindrem, 2026-02-01:
A note on the new RPI touchscreen:

If I recall, the original RPI touchscreen works "out of the box" without using RpiDisplaySetup so maybe this issue should be raised on the Victron forum.

Possible issues:

The installation manual for the RPI touchscreen 2 mentions a device tree overlay but seems to be optional.

There is also mention of video=DSI-1 but again this appears to be optional.

The manual recommends choosing the DSI port labeled 1. Might try DSI 0.

MTUpower, 2026-02-02:
Hi Kwindrem, and many thanks for the reply. Unfortunately I have no clue what you are talking about re DSI etc or even how or what to do with them not being computer savvy.  But will look at the Victron Forum and see what that's about.  But if you do happen to stumble upon the solution I'm sure a few people would be much appreciative. Thank you