kwindrem/RpiDisplaySetup #44: MPI3508 screen - no touch control

jaeger81, 2025-01-29:
Hello,
I tried to install 3,5" screen with GPIO and HDMI interfaces using RpiDispalySetup but I only managed to make it work with HDMI (so no touch control). Tried everything possible solution I found around but non worked. Is there any solution to this issue? While trying different solutions I found that when I want to run ts_calibrate I get: ts_setup: No such file or directory. I am absolutely a newbie to this. Maybe it's missing driver?

jaeger81, 2025-01-30:
> Not all HDMI touchscreens will work with Venus OS and there are far too many variations to guess at the cause.
> The display needs to provide touch information via USB. There is no GPIO interface for touch.
> ts_calibrate is not included in the stock firmware. RpiDisplaySetup attempts to install it from the internet if you request calibration from the setup script. I can't say why what you are doing isn't working without logs and more information about what steps you have taken.

Thanks for reply! I really appreciate it. Maybe it would be easiest if you tell me if you know any of the 3,5” displays that work seamlessly with your script🙂Regards,Michael

kwindrem, 2025-01-30:
Sorry, no I don't have any experience with smaller displays

jaeger81, 2025-01-31:
That’s a pity 🤷‍♂️ maybe I’ll try to figure it out with the one I have. Question: I have both possibilities to connect it to too: Gpio and or hdmi plus usb. Should I connect all the interfaces (this seems not to work) or stick only to hdmi/usb?And if were to help me what logs should I attach?Regards,Michael

kwindrem, 2025-01-31:
If you mean DSI (not GPIO), this is the "native" display interface for the PI. I can't speak to which would be better.

All logs are in /data/log. You might have a look at messages there. It's where some of the boot messages are. There is also the command line dmesg which gives other issues.

Look at the documentation for the display in question. If it requires installation of a special driver, you may not be able to make it work with Venus OS.

scumari, 2025-12-28:
Hi all, Have a 3,5 inch HDMI display with MPI3508 working with KlipperScreen.
First create a RasberryPi OS Lite 64bit image, then install LCD-show.
Then install kiauh and with this install all needed.