kwindrem/RpiDisplaySetup #46: RpiDisplaySetup not work beta 57 / 58

kurt1712, 2025-04-12:
Hello, since the last beta 3.60b57/58 of venusOS I have no more monitor output (Pi4). Booting ends in the command prompt. A new installation of RpiDisplaySetup did not help either. Is this a known problem or my problem :( ? What can I do ? Thanks for your help

kwindrem, 2025-04-12:
Yes, this is a known issue that I'm working on. Victron has made some changes in the way the local display is handled and with early v3.60 betas it looked like they were moving towards always running the local display on the PI platforms. With more recent betas it appears they have backed off on this.

I will release a change to RpiDisplaySetup that goes back to the old logic which should restore the local GUI on the PI after running the setup script.

If you want to do something before then, you can remove the headless file and reboot:

rm /etc/venus/headless
reboot

kwindrem, 2025-04-12:
I've released v3.16 of RpiDisplaySetup that restores the use of the headless flag. This should allow the local display to go active in the more recent firmware v3.60 betas

closed 2025-04-12 by kwindrem