kwindrem/RpiDisplaySetup #47: RPi Touchscreen - Blank

jaket91-1, 2025-05-12:
Hi,

Firstly thanks for all your work on these projects!

I have an original Raspberry Pi 7'' Touchscreen Display connected to a Rpi4B as well as a waveshare CAN hat.

Some time last year (unfortunately I can’t remember the exact change/release) the screen stopped working and is blank. It was working great prior, I think it was when the “New UI” came out.

I’d like to get it working again so I’m revisiting it. I’ve upgraded VenusOS and reinstalled SetupHelper and RPIDisplaySetup but to no avail. At boot the display shows the RPi test colour square and then the Victron splash screen but then goes black and is unresponsive.

My VenusOS is version 3.60~72 and SetupHelper is on v8.33 and RPIDisplaySetup is v3.16. 

The calibration seems to hang whenever running the setup on the CLI, not sure if that’s relevant.

Is it possible the CAN hat is interfering now but wasn’t previously?

Any suggestions on what I can do further to get it working would be greatly appreciated.

kwindrem, 2025-05-12:
I have the same setup for my test system and it is working fine. I just upgraded to v3.60~72 and that is also working.

I would suggest running /data/RpiDisplaySetup/setup from the command line and resetting all the configuration (a full install, not reinstall).

the ts_calibrate routine can not be run from the command line without setting /etc/venus/headless and rebooting. Running from RpiDisplaySetup does this for you and I just tested this and it works as designed. Note that the touchscreen calibration won't work if you can't see the screen and should have no bearing on showing an image on the screen.

Not sure of the CANbus hat would interfere now if it was not before but you can always pull it off and try.

You might check the power and DSI connections to the display. I've seen mine go blank with a brief power interruption. After that happens a reboot is needed but it always comes back (assuming power has been reconnected).

Not sure what else I have to contribute.

jaket91-1, 2025-05-13:
Thanks for the reply! Would you mind sharing the relevant parts of your `config.txt`, please? I've removed the CAN hat and get exactly the same result. Clearly the display is working because I see the splash screen, which leads me to believe there's probably a very simple config option that needs flipped. I only have the power pins connected which looks to be correct from the articles I've read.

kwindrem, 2025-05-13:
The local display will not appear if /etc/venus/headless flag file exists. But in this case, I get the Victron splash screen, not blank.